ros2/rmw #13: specified queue size in rmw interface

dirk-thomas, 2015-05-19:
Currently it is only ensured that the history depth of the data reader and writer is *at least* the specified size.

While this allow the user to use vendor specific configuration to override the QoS parameters it prevents the user from setting a not-to-be-changed queue size.

We might want to consider offering both options to the developer and decide which behavior should be the default.

jacquelinekay, 2015-06-30:
To be considered in https://github.com/ros2/design/pull/36

wjwwood, 2018-02-22:
@dirk-thomas is this still an issue? Do you remember why depth is only "ensured" and not enforced?

dirk-thomas, 2018-02-22:
I have no knowledge of a change in the behavior regarding this so I expect it to be still the same issue.

wjwwood, 2018-02-22:
Can answer my second question? Why is it this way in the first place? Is it because of the global default, driven by the dds configuration files?

dirk-thomas, 2018-02-22:
Because that is how it is currently implemented: https://github.com/ros2/rmw_connext/blob/e0c06416625153ecb0e56b115dc2d90da3646841/rmw_connext_shared_cpp/include/rmw_connext_shared_cpp/qos.hpp#L95-L107

wjwwood, 2018-02-22:
I understand that this is how it's currently implemented, I was asking _why_ was it implemented that way? I was asking if there was a good reason for it, or if it was that this alternative was picked at random. Or if you were aware of any discussion where we discussed the two alternatives.

I finally managed to find the original conversation about this (it was in a collapsed inline comment thread): https://github.com/ros2/rmw_opensplice/pull/31#discussion_r30633851

The summary is basically that there needs to be a way to either specify exactly what you want or use the system default and/or optionally allow for the current behavior which is request a history depth, but use the system default if it's bigger.