ros2/rmw #38: RMW_CHECK_TYPE_IDENTIFIERS_MATCH macro for type support

jacquelinekay, 2015-09-08:
Currently, the macro `RMW_CHECK_TYPE_IDENTIFIERS_MATCH` has an error message of the format:

`" implementation '%s'(%p) does not match rmw implementation '%s'(%p)"`

However, in the rmw implementations this macro is used for checking the types of typesupport identifiers, not only rmw implementation identifiers, leading to confusing and incorrect error messages such as:

`could not create publisher: type support handle implementation 'connext_static'(0x2b54f95e9671) does not match rmw implementation 'introspection'(0x2b54f97eb691)`

This macro should be reworded to be more general, or a different macro should be made for matching typesupport identifiers.

tfoote, 2016-05-03:
@jacquelinekay to update status on this issue/pr.

dirk-thomas, 2017-04-10:
I don't see any cases anymore where this macro is being invoked with type support arguments. Therefore I will close this.

closed 2017-04-10 by dirk-thomas