ros2/rmw #77: Potentially risky to use transient_local durability for services

dhood, 2016-11-11:
@wjwwood and I were talking yesterday and realised that the default QoS profile for services might lead to undesired behaviour. 

The default durability [is `RMW_QOS_POLICY_TRANSIENT_LOCAL_DURABILITY`](https://github.com/ros2/rmw/blob/a7436c86d477665c38e0122ef71598835d6560f1/rmw/include/rmw/qos_profiles.h#L54). @wjwwood pointed out that this was set before `wait_for_service` was implemented, as a way to compensate for race conditions in server/client bringup.

With this setting, in the case of a server that starts, dies, and restarts, it will receive the service request from the client twice. **While the client is protected from receiving multiple service responses, the server is not protected from side effects of receiving multiple service calls.**

If a service does anything like write to file, increment a counter, toggle a value, etc, the transient_local setting is probably not a good idea.

A better default might be volatile, and we should be careful about exposing the transient_local setting to users.

wjwwood, 2016-11-11:
+1 to all of this.

I'd actually like to either not expose this QoS setting in the user facing Service interface or otherwise somehow enforce that they do not use something other than volatile. Simply because it will still work, but have this subtle and unsafe behavior.

--

As a side anecdote, we toggled our blade state with a Service call, having a duplicate request (even if the response was discarded) would have been a terrible surprise.

wjwwood, 2016-11-11:
Err, I should have said "on our autonomous lawnmower, we toggled our blade state..."

dhood, 2016-12-14:
closed with #79

closed 2016-12-14 by dhood