ros2/rmw #85: Inconsistent naming in QoS policy options

dhood, 2016-12-11:
We have a lot of different formats for the names of [QoS policy settings](https://github.com/ros2/rmw/blob/07c605b3561aa5bb82744dc236b574ac7e42b879/rmw/include/rmw/types.h#L155..L174) at the moment. The inconsistency makes it inconvenient for users to change the options without looking them up.

If we unify the names, what should the format be?

1. `RMW_QOS_POLICY_{SETTING}_{CATEGORY}` e.g. how we have RMW_QOS_POLICY_KEEP_LAST_HISTORY (closest to DDS naming e.g. RELIABLE_RELIABILITY_QOS)
2. `RMW_QOS_POLICY_{CATEGORY}_{SETTING}` e.g. how we have RMW_QOS_POLICY_HISTORY_SYSTEM_DEFAULT
3. `RMW_QOS_POLICY_{SETTING}` e.g. how we have RMW_QOS_POLICY_RELIABLE
4. `RMW_QOS_{CATEGORY}_POLICY_{SETTING}` e.g. how we have rmw_qos_history_policy_t

or perhaps a different (shorter) option

mikaelarguedas, 2016-12-11:
If we keep a verbose way of naming them I'd vote for 2 that makes it clearer for human and regex to find what parameter it's impacting. Otherwise a shorter version could be convenient for readability and typing.

It comes down to how often do we expect user to change many QoS fields versus using the predefined profiles and changing only one parameter from time to time

gbiggs, 2016-12-12:
The second option seems the most readable, but as @mikaelarguedas says it is quite long to type.

dhood, 2016-12-13:
thanks @mikaelarguedas and @gbiggs 
@ros2/team looking for opinions please

dirk-thomas, 2016-12-13:
Option 2. sounds good to me.

wjwwood, 2016-12-13:
Option 2.

mikaelarguedas, 2016-12-13:
all associated PRs look good to me

closed 2016-12-13 by dhood