ros2/rmw #113: Plumb PERSISTENT durability into RMW

clalancette, 2017-07-14:
I don't know the full scope of doing this, but I know for sure that in order to get persistent durability, we'll need to add a new member to the durability_policy_t here: https://github.com/ros2/rmw/blob/master/rmw/include/rmw/types.h#L191 .  I'm sure there is other work to get it down to the RMW implementation properly.

dirk-thomas, 2017-07-14:
We don't need `persistent` in order to mimic latching support. Actually we decided not to use the `persistent` durability since it extends the data lifetime beyond the lifetime of the entity which published it.

mikaelarguedas, 2017-07-19:
Based on feedback and given that https://github.com/eProsima/Fast-RTPS/issues/120 has been closed I/m going to close this

closed 2017-07-19 by mikaelarguedas