ros2/rmw #115: Expose more QoS settings

mikaelarguedas, 2017-08-04:
Meta-ticket to track the progress in exposing more QoS settings. 
Please update the tasklist below when a setting has been decided to be exposed.

- [ ] 
- [ ]

dirk-thomas, 2017-08-04:
The short term goal is to expose the underlying DDS handles to allow users to access anything they need (for the downside of locking themselves into the API of a specific vendor).

mikaelarguedas, 2017-10-16:
An example of using DDS handles has been provided in https://github.com/ros2/demos/pull/164
Since https://github.com/ros2/rmw_fastrtps/pull/153 It is also now possible to override vendor defaults QoS settings in fastrtps as is was the case for connext

wjwwood, 2018-02-22:
Closing in favor of more specific issues, one for each additional QoS we want to expose.

closed 2018-02-22 by wjwwood