ros2/rmw #183: Modify implementations to use one Participant per Context

ivanpauno, 2019-09-03:
## Feature request

#### Feature description

Design document [here](https://github.com/ros2/design/pull/250).
The idea is to implement everything in the `rmw` layer, and avoid changes in upstream layers as much as possible.

Here a list of tasks, with a rough estimation of task's duration:

1. [ ] Generate message interfaces in a common package, that can be built before the rmw implementations and used from them. [3 days FTE]
2. [ ] Use group data in publishers and subscribers for communicating the node name, and create a topic cache with it. [3 days FTE]
  Similar to https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/topic_cache.hpp.
It'll be a common implementation of the topic cache (probably, with some implementation specific details provided by the specific rmw implementation, like the guid data type used).
  It will be organized in the following way:
    - A map where the keys are the topic names, and the values the types (TopicToTypes)
    - A map from node_name to TopicToTypes map.
    - A map from participant guid to node names.
  The implementation of the ParticipantListener that reads this data is specific of each rmw implementation, though some helper functions could be available in the common package.
3. [ ] For recognizing services/clients, an approach similar to https://github.com/ros2/rmw_fastrtps/blob/099f9eed9a0f581447405fbd877c6d3b15f1f26e/rmw_fastrtps_shared_cpp/src/rmw_node_info_and_types.cpp#L392-L502 will be used.
  Basically, services/clients are recognized by a custom prefix and collected from the same topic cache. This could be provided in the common implementation. [2 day FTE]
4. [ ] Publish a message each time a node is created with the updated list of nodes of each participant. [1 day FTE]
5. [ ] Implement a `node_listener` thread, which subscribes to that topic and updates a map from participant guid to a vector of node names [2 days FTE].
6. [ ] Modify Context/Node/Publisher/Subscriber/Client/Service creation and deletion functions as needed [4 days FTE].
7. [ ] Update graph functions to access this data [4 days FTE].
8. [ ] Introduce the concept of context in policy/governance/access_control_permission files. [5 days FTE] (rough guess, I don't have enough context).
9. [ ] Update sros2 command line tool. [3 days FTE].

Steps 1, 3, 4, 5, 7, 8, 9 could be provided in a common package for all the rmw implementations.
The `topic cache` implementation in step 2 too, but each rmw implementation will have its own ParticipantListener. The part that is needed to be repeated should take 1 day FTE.
Step 6 should be repeated for each rmw implementation.

This results in:
- 27 days FTE for updating the first rmw implementation.
- 5 extra days for each extra rmw implementation updated.

----

The only blocker to avoid modifying all the rmw implementations in one step, is that sros2 API will change, and it's currently provided in rcl with a common implementation.
To avoid that, the following can be done:
- Move security API from rcl to rmw.
- Use the old behavior "file formats"/"directory tree" for implementations that haven't been modified, and the new format in the ones who do.
- sros2 command line tool will check `RMW_IMPLEMENTATION` environment variable, and behave differently depending on it.

In the meantime we will be breaking cross-vendor tests, so they should be disabled.

----

The task can be easily parallelized in:
- "discovery" related changes. [19 days FTE]
- sros2 related changes. [8 days FTE]

ivanpauno, 2019-09-03:
@hidmic @dirk-thomas @wjwwood FYI

ivanpauno, 2019-12-18:
Linked PRs:
rmw_fastrtps: https://github.com/ros2/rmw_fastrtps/pull/312
rmw: https://github.com/ros2/rmw/pull/189
rmw_implementation: https://github.com/ros2/rmw_implementation/pull/77
rcl: https://github.com/ros2/rcl/pull/515
rmw_dds_common (new package): https://github.com/ivanpauno/rmw_dds_common/pull/4/

Other related PRs:
https://github.com/ros2/rclpy/pull/470

I will open PRs in others rmw implementations with minimal API changes, though they will be refactored in a second stage.

ivanpauno, 2020-03-03:
I've updated the ticket with more up to date information.

clalancette, 2021-03-29:
I'm fairly sure this is done now.  At least, I know for sure that it is fixed in Fast-DDS and Connext.  I'm also fairly certain this is the case for Cyclone DDS.  @eboasson @ivanpauno Please feel free to reopen if I'm wrong about Cyclone.

closed 2021-03-29 by clalancette