ros2/rmw #184: ASAN: heap over-read near rmw::impl::cpp:parse_key_value

oneattosecond, 2019-09-10:
## Bug report

Seeing a heap over-read reported by ASAN when running cartographer or navigation on MacOS (compiled with ASAN enabled) 

**Required Info:**

- Operating System:
  - MacOS 10.16.4
- Installation type:
  - compiled from source
- Version or commit hash:
  - latest dashing .repos file
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - reproduces in cartographer and navigation modules, multiple nodes

#### Steps to reproduce issue
1. enable ASAN by adding below to cartographer and navigation CMakeLists.txt files

    set (CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG} -fno-omit-frame-pointer -fsanitize=address")
    set (CMAKE_LINKER_FLAGS_DEBUG "${CMAKE_LINKER_FLAGS_DEBUG} -fno-omit-frame-pointer -fsanitize=address")

2. recompile cartographer, navigation (didn't try recompiling all of ROS with ASAN, not sure if that's a mistake)

    colcon build --cmake-args -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Debug -DFORCE_DEBUG_BUILD=True --symlink-install

3. run cartographer, navigation as normal

4. observe ASAN flagging 8-byte heap over-read

#### Expected behavior
No traps from ASAN

#### Actual behavior
ASAN flags over-read and dies

#### Additional information

I've reviewed a few different nodes, and they all have some form of this stack trace, where a node (multiple of them are killed by ASAN) is doing something that invokes rmw::impl::cpp:parse_key_value()

[occupancy_grid_node-2] =================================================================
[occupancy_grid_node-2] ==37032==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000042d5 at pc 0x0001034d46df bp 0x70000b8f0c30 sp 0x70000b8f03e0
[occupancy_grid_node-2] READ of size 8 at 0x6020000042d5 thread T3
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2]     #0 0x1034d46de in __asan_memcpy (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x596de)
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2] [WARN] [occupancy_grid_node]: submap_slices and last_frame_id is empty
[occupancy_grid_node-2]     #1 0x100906a8c in std::__1::enable_if<((std::__1::integral_constant<bool, true>::value) || (!(__has_construct<std::__1::allocator<unsigned char>, bool*, bool>::value))) && (is_trivially_move_constructible<bool>::value), void>::type std::__1::allocator_traits<std::__1::allocator<unsigned char> >::__construct_backward<unsigned char>(std::__1::allocator<unsigned char>&, bool*, bool*, bool*&) memory:1699
[occupancy_grid_node-2]     #2 0x10090550a in std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >::__swap_out_circular_buffer(std::__1::__split_buffer<unsigned char, std::__1::allocator<unsigned char>&>&) vector:931
[occupancy_grid_node-2]     #3 0x101aee75a in void std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >::__push_back_slow_path<unsigned char const&>(unsigned char const&&&) (librclcpp.dylib:x86_64+0xdc75a)
[occupancy_grid_node-2]     #4 0x111460d93 in std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >::push_back(unsigned char const&) (librmw_fastrtps_shared_cpp.dylib:x86_64+0x21d93)
[occupancy_grid_node-2]     #5 0x11146005a in rmw::impl::cpp::parse_key_value(std::__1::vector<unsigned char, std::__1::allocator<unsigned char> >) (librmw_fastrtps_shared_cpp.dylib:x86_64+0x2105a)
[occupancy_grid_node-2]     #6 0x11145d855 in ParticipantListener::onParticipantDiscovery(eprosima::fastrtps::Participant*, eprosima::fastrtps::rtps::ParticipantDiscoveryInfo&&) (librmw_fastrtps_shared_cpp.dylib:x86_64+0x1e855)
[occupancy_grid_node-2]     #7 0x103057aa6 in eprosima::fastrtps::rtps::PDPSimpleListener::onNewCacheChangeAdded(eprosima::fastrtps::rtps::RTPSReader*, eprosima::fastrtps::rtps::CacheChange_t const*) (libfastrtps.1.dylib:x86_64+0x1c8aa6)
[occupancy_grid_node-2]     #8 0x102ee419b in eprosima::fastrtps::rtps::StatelessReader::change_received(eprosima::fastrtps::rtps::CacheChange_t*) (libfastrtps.1.dylib:x86_64+0x5519b)
[occupancy_grid_node-2]     #9 0x102ee4372 in eprosima::fastrtps::rtps::StatelessReader::processDataMsg(eprosima::fastrtps::rtps::CacheChange_t*) (libfastrtps.1.dylib:x86_64+0x55372)
[occupancy_grid_node-2]     #10 0x102ef3a8c in eprosima::fastrtps::rtps::MessageReceiver::proc_Submsg_Data(eprosima::fastrtps::rtps::CDRMessage_t*, eprosima::fastrtps::rtps::SubmessageHeader_t*) (libfastrtps.1.dylib:x86_64+0x64a8c)
[occupancy_grid_node-2]     #11 0x102ef1869 in eprosima::fastrtps::rtps::MessageReceiver::processCDRMsg(eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::CDRMessage_t*) (libfastrtps.1.dylib:x86_64+0x62869)
[occupancy_grid_node-2]     #12 0x102efaaac in eprosima::fastrtps::rtps::ReceiverResource::OnDataReceived(unsigned char const*, unsigned int, eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::Locator_t const&) (libfastrtps.1.dylib:x86_64+0x6baac)
[occupancy_grid_node-2]     #13 0x102f57a91 in eprosima::fastrtps::rtps::UDPTransportInterface::perform_listen_operation(eprosima::fastrtps::rtps::UDPChannelResource*, eprosima::fastrtps::rtps::Locator_t) (libfastrtps.1.dylib:x86_64+0xc8a91)
[occupancy_grid_node-2]     #14 0x102f5b402 in void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, void (eprosima::fastrtps::rtps::UDPTransportInterface::*)(eprosima::fastrtps::rtps::UDPChannelResource*, eprosima::fastrtps::rtps::Locator_t), eprosima::fastrtps::rtps::UDPTransportInterface*, eprosima::fastrtps::rtps::UDPChannelResource*, eprosima::fastrtps::rtps::Locator_t> >(void*) (libfastrtps.1.dylib:x86_64+0xcc402)
[occupancy_grid_node-2]     #15 0x7fff5c05a2ea in _pthread_body (libsystem_pthread.dylib:x86_64+0x32ea)
[occupancy_grid_node-2]     #16 0x7fff5c05d248 in _pthread_start (libsystem_pthread.dylib:x86_64+0x6248)
[occupancy_grid_node-2]     #17 0x7fff5c05940c in thread_start (libsystem_pthread.dylib:x86_64+0x240c)
[occupancy_grid_node-2] 
[occupancy_grid_node-2] 0x6020000042d8 is located 0 bytes to the right of 8-byte region [0x6020000042d0,0x6020000042d8)
[... 66 lines cut ...]
[occupancy_grid_node-2]   0x1c0400000870: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
[occupancy_grid_node-2]   0x1c0400000880: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
[occupancy_grid_node-2]   0x1c0400000890: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
[occupancy_grid_node-2]   0x1c04000008a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
[occupancy_grid_node-2] Shadow byte legend (one shadow byte represents 8 application bytes):
[occupancy_grid_node-2]   Addressable:           00
[occupancy_grid_node-2]   Partially addressable: 01 02 03 04 05 06 07 
[occupancy_grid_node-2]   Heap left redzone:       fa
[occupancy_grid_node-2]   Freed heap region:       fd
[occupancy_grid_node-2]   Stack left redzone:      f1
[occupancy_grid_node-2]   Stack mid redzone:       f2
[occupancy_grid_node-2]   Stack right redzone:     f3
[occupancy_grid_node-2]   Stack after return:      f5
[occupancy_grid_node-2]   Stack use after scope:   f8
[occupancy_grid_node-2]   Global redzone:          f9
[occupancy_grid_node-2]   Global init order:       f6
[occupancy_grid_node-2]   Poisoned by user:        f7
[occupancy_grid_node-2]   Container overflow:      fc
[occupancy_grid_node-2]   Array cookie:            ac
[occupancy_grid_node-2]   Intra object redzone:    bb
[occupancy_grid_node-2]   ASan internal:           fe
[occupancy_grid_node-2]   Left alloca redzone:     ca
[occupancy_grid_node-2]   Right alloca redzone:    cb
[occupancy_grid_node-2]   Shadow gap:              cc
[occupancy_grid_node-2] ==37032==ABORTING

oneattosecond, 2019-09-11:
cool I'll try that today, time willing.

oneattosecond, 2019-09-11:
with TB3 teleop node running:

[occupancy_grid_node-2] name=teleop_keyboard;namespace=/;
[occupancy_grid_node-2] =================================================================
[occupancy_grid_node-2] ==58014==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000052d5 at pc 0x00010fd4a6df bp 0x7000002ddc00 sp 0x7000002dd3b0
[occupancy_grid_node-2] READ of size 8 at 0x6020000052d5 thread T3

without teleop running:

[occupancy_grid_node-2] name=rviz2;namespace=/;
[occupancy_grid_node-2] name=launch_ros;namespace=/;
[occupancy_grid_node-2] =================================================================
[occupancy_grid_node-2] ==58033==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000045d5 at pc 0x0001038ea6df bp 0x700002fa1c00 sp 0x700002fa13b0
[occupancy_grid_node-2] READ of size 8 at 0x6020000045d5 thread T3

oneattosecond, 2019-09-11:
on navigation launch, few different signatures

[world_model-4] name=bt_navigator;namespace=/;
[world_model-4] =================================================================
[world_model-4] ==58132==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000052f5 at pc 0x00010ff896df bp 0x700005fe2c00 sp 0x700005fe23b0
[world_model-4] READ of size 8 at 0x6020000052f5 thread T3

[lifecycle_manager-1] name=map_server;namespace=/;
[lifecycle_manager-1] =================================================================
[lifecycle_manager-1] ==58129==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000042f5 at pc 0x0001072e56df bp 0x700001c82c00 sp 0x700001c823b0
[lifecycle_manager-1] READ of size 8 at 0x6020000042f5 thread T3

[recoveries_node-7] name=launch_ros;namespace=/;
[recoveries_node-7] =================================================================
[recoveries_node-7] ==58135==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000052f5 at pc 0x00010720d6df bp 0x70000306bc00 sp 0x70000306b3b0
[recoveries_node-7] READ of size 8 at 0x6020000052f5 thread T3

[dwb_controller-5] name=rviz2;namespace=/;
[dwb_controller-5] name=launch_ros;namespace=/;
[dwb_controller-5] =================================================================
[dwb_controller-5] ==58133==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000055f5 at pc 0x00011097e6df bp 0x70000dba0c00 sp 0x70000dba03b0
[dwb_controller-5] READ of size 8 at 0x6020000055f5 thread T3

clalancette, 2021-06-24:
This is an old issue, and Dashing is now End-of-Life.  We've also fixed a lot of bugs in this area.  So I'm going to close this one out.  If you run across this again with a newer ROS distribution, please feel free to open another issue.

closed 2021-06-24 by clalancette