ros2/rmw #197: why not definite it in the rmw.h

reed-lau, 2020-02-27:
https://github.com/ros2/rmw/blob/28f91855510c3c832b500fae40fdcc4169d26a0d/rmw/include/rmw/event.h#L16

here the event-related function is declared in the `event.h` header file, but it is not included in the rmw.h.
when we implement a rmw-xxx, we have to include another rmw/event.h, why the `event` is so special.

dirk-thomas, 2020-02-28:
I would rather ask the opposite question: why should everything be declared in a single header? The fact that this single header contains numerous unrelated function declarations and is 1000+ lines long would instead encourage me to split it up into separate files - each containing only functions of a specific category (e.g. pub/sub in one, services in another, etc.).

reed-lau, 2020-02-28:
> each containing only functions of a specific category

that sounds great!
I just mean, `event` is not special enough.