ros2/rmw #206: Progress toward Quality Level 1

brawner, 2020-03-25:
This issue tracks the progression of `rmw` to Quality Level 1 and a 1.0 version level. It follows the outline described in [REP 2004](https://github.com/ros-infrastructure/rep/blob/c2ba03bac23b8f313e47820833fad000dbbe058b/rep-2004.rst). 

* [x] Version Policy
    - Follows ROS Core Quality Declaration
* [ ] Version >= 1.0.0
* [x] Change Control Process
    - Follows ROS Core
* [ ] PRs automatically tested on Jenkins

Documentation
* [x] Per-feature documentation. 
    - Features still needing documentation on README.md or equivalent (`endian.hpp`, `filesystem_helper.hpp`, `find_and_replace.hpp`, `find_library.hpp`, `join.hpp`, `split.hpp`, `pointer_traits.hpp`, `visibility_control.hpp`)
* [x] Per-item documentation in public API. 
    - Items need doc blocks (`endian.hpp`, `filesystem_helper.hpp`, `thread_safety_annotations.hpp`, `visibility_control.hpp`)
    - Items needing doc improvements (`join.hpp`)
* [x] Declared set of licenses
* [x] Copyright statement in each source file
* [x] Quality Declaration document (#205)

Testing:
* [x] System tests, which cover "features" (#203)
* [x] Tests covering public API (#203)
* [x] Code coverage (with #203)
* [ ] Performance tests
* [x] Performance test policy: ROS Core
* [x] Package code style: ROS Core
* [x] Uses ament_lint_common

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
    - rcutils
* [x] non-ROS dependencies are equivalent level 1
    - c++ standard library

Platform Support
* [x] Supports all tier 1 platforms

jacobperron, 2020-05-12:
We now have a 1.0.0 release: https://github.com/ros/rosdistro/pull/24838

chapulina, 2020-11-25:
This package reached level 1! https://github.com/ros2/rmw/pull/289

closed 2020-11-25 by chapulina