ros2/rmw #226: Add support for `on sample lost` event

ivanpauno, 2020-05-06:
## Feature request

#### Feature description

Similar to previously added events like https://github.com/ros2/ros2/issues/822 or `on liveliness lost`, it would be good to add support to `on sample lost` events (which might be really helpful when debugging communication issues).

For `ROS`, `on message lost` might be a more appropriate name.

For DDS based rmw implementations, implementing this should be pretty straightforward:
- Connext: [sample lost status](https://community.rti.com/static/documentation/connext-dds/5.2.3/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/Statuses_for_DataReaders.htm#receiving_2076951295_607362), [data reader listener callbacks](https://community.rti.com/static/documentation/connext-dds/5.2.3/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/Setting_Up_DataReaderListeners.htm#receiving_2076951295_426659).
- FastRTPS: AFAIS, it's not supported yet: https://github.com/eProsima/Fast-RTPS/blob/8842bdd35bac66b731fe3f2f5e79fe7417355c4b/include/fastdds/rtps/reader/ReaderListener.h#L41.
- CycloneDDS: https://github.com/eclipse-cyclonedds/cyclonedds/blob/623b35d83b9d061187c816f37f4a92798f3b35d1/src/core/ddsc/include/dds/ddsc/dds_public_listener.h#L37.

#### Implementation considerations

For rmw implementations that don't support the feature, they can return [RMW_RET_UNSUPPORTED](https://github.com/ros2/rmw/blob/10bb863176fcd832739eed3077d2fcbe9d619f64/rmw/include/rmw/ret_types.h#L34), as for example fastrtps already does when the creation of an `ON_*_INCOMPATIBLE_QOS` event is requested.

DDS also support a status callback called [on sample rejected](https://community.rti.com/static/documentation/connext-dds/5.2.3/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/Statuses_for_DataReaders.htm#receiving_2076951295_607403). The difference with the previous one, is that this one reflects a reader not having resources to receive the sample and not a communication problem. I think it would be interesting to add support to that one too, maybe both can be wrapped in the same event?

ivanpauno, 2020-05-06:
> maybe both can be wrapped in the same event?

That sound like possible, the status definition could be:

```c
typedef struct RMW_PUBLIC_TYPE rmw_message_lost_status_t
{
  size_t total_count; // total number of messages lost
  size_t total_count_change; // number of messages lost since last callback
  char * reason; // reason why the message was lost/rejected
} rmw_liveliness_changed_status_t;
```

Based on both DDS status definitions ([1](https://community.rti.com/static/documentation/connext-dds/5.2.3/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/Statuses_for_DataReaders.htm#receiving_2076951295_607362), [2](https://community.rti.com/static/documentation/connext-dds/5.2.3/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/Statuses_for_DataReaders.htm#receiving_2076951295_607403)), we can use that to wrap both `on sample lost` and `on sample rejected`.

ivanpauno, 2020-05-06:
@jacobperron @clalancette @wjwwood for feedback (no urgency right now, as an implementation wouldn't start until `Foxy` gets released).

fujitatomoya, 2020-05-07:
This is really helpful for our use case. we can expect and accept message drops sometimes with wireless network for edge devices, but what's better is user application is likely to be notified if message drops.

fujitatomoya, 2020-06-08:
different topic, but we'd also really like to see this in statistics for actual use case.

ivanpauno, 2020-06-08:
Opened PRs:
* [rmw](https://github.com/ros2/rmw/pull/232)
* [rmw_connext](https://github.com/ros2/rmw_connext/pull/424)
* [rmw_cyclonedds](https://github.com/ros2/rmw_cyclonedds/pull/192)
* [rcl](https://github.com/ros2/rcl/pull/673)

Missing PRs:
* rmw_fastrtps (does it have support?)
* rclcpp
* rclpy

jacobperron, 2020-06-09:
I'm not convinced that we want to combine "sample lost" and "sample rejected" based on the discussion in https://github.com/ros2/rmw_cyclonedds/pull/192#issuecomment-641070349. It'd be nice to have more DDS-inclined folks chime in.

ivanpauno, 2020-06-09:
> I'm not convinced that we want to combine "sample lost" and "sample rejected" based on the discussion in [ros2/rmw_cyclonedds#192 (comment)](https://github.com/ros2/rmw_cyclonedds/pull/192#issuecomment-641070349). It'd be nice to have more DDS-inclined folks chime in.

Yes, I agree.
I will address that comment soon.

ivanpauno, 2020-07-14:
Closing, as all tasks have been done.
The only remaining item is a `rmw_fastrtps` implementation, tracked by https://github.com/ros2/rmw_fastrtps/issues/395.

closed 2020-07-14 by ivanpauno