ros2/rmw #231: Asynchronously shutting down while using `rclcpp::spin` `rclcpp:spin_some` in another thread throws an exception

ivanpauno, 2020-05-27:
This nightly failure is an example of the problem:

- https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/1968/testReport/junit/(root)/projectroot/test_lifecycle_service_client/

What is happening is that one thread is creating an executor, while other is trying to shutdown.
If shutdown is called before the creation of the `Executor`, an exception is thrown:

https://github.com/ros2/rclcpp/blob/87bb9f9758dce239d37256ae91d58ce6325f405b/rclcpp_lifecycle/test/test_lifecycle_service_client.cpp#L218-L231

ivanpauno, 2020-05-27:
@hidmic @wjwwood @jacobperron does this look like an user error (i.e. the test is not well written) or should be improve how shutdown works?

hidmic, 2020-05-27:
I'd say user error.

jacobperron, 2020-05-27:
Seems like user error; calling shutdown before spin. What's an alternative, have spin return if the context is invalid?

ivanpauno, 2020-05-27:
> Seems like user error; calling shutdown before spin. What's an alternative, have spin return if the context is invalid?

The thing is that `spin_some` is called in a loop, "protected" in a `rclcpp::ok` check.
If the context is shutdown in the middle of the call to `rclcpp::ok` and the construction of the executor in `rclcpp::spin_some`, then the error happens.

IMO, it's a behavior error, and not an user one (I would expect that if I do something after checking for `rclcpp::ok`, there should be no problem).

IMO shutdown shouldn't affect executors/publishers/subscriptions/nodes/... etc, it should only make `rclcpp::ok()` return false and wake up executors.

In that way, we could shutdown asynchronously without this error (or other bunch of possible of similar errors).

wjwwood, 2020-05-27:
I also would classify it as "user error", but not necessarily the user's fault. It's confusing and so the existing story could be improved.

This has been discussed in the past:

- https://github.com/ros2/rclcpp/issues/812
- https://github.com/ros2/rclcpp/issues/610
- https://github.com/ros2/ros2/issues/369#issuecomment-518779552

And perhaps others, though I couldn't find them. I know @sloretz and I spoke about it in person at length around the time https://github.com/ros2/rmw/pull/154 was merged.

And this might be a duplicate?

- https://github.com/ros2/rclcpp/issues/1111

----

Originally I thought about shutdown like KeyboardInterrupt in Python, but since this is C++ and the user isn't always inside on of our functions we cannot ensure it is raised everywhere. Instead we have (some of ) our functions raise and then the user can check `rclcpp::ok()` to know when to interrupt their own work.

The thinking was that if the user wrote something like this:

```c++
while (true) {
  rclcpp::spin_some(...);
}
```

Then they wouldn't get stuck in a loop due the mistake (imo) of not checking `rclcpp::ok()`. I think this fail fast approach will result in the fewest unexplained hangs for users.

However, that means to avoid tracebacks and crashes they need to catch and handle these errors (like you do with KeyboardInterrupt in Python), and it's also hard to know which things will work and which will not. For example, publishing will work, and destroying a publisher will work, but making a new publisher will not. This was essentially a compromise to enable things like "during shutdown I might want to send a message (publish) to notify someone", for example.

The rules not being clear on this is something that Apex folks struggled with, and it's clearly been an issue for us and our users. So I'd be open to reevaluating this decision. We could instead say that shutdown changes nothing but a flag in the context, and everything continues as normal and it's up to the user to notice and react to it.

wjwwood, 2020-05-27:
> IMO, it's a behavior error, and not an user one (I would expect that if I do something after checking for rclcpp::ok, there should be no problem).

This is a bad assumption, imo. It's like acquiring a lock, releasing it, and then assuming afterwards the resource is safe. I understand how it's an easy mistake to make, but imo it is a mistake. `rclcpp::ok()` just checks a state, it doesn't hold a resource or block other actions, and it doesn't indicate that to be the case.