ros2/rmw #234: test_names_and_types failing on Linux

nuclearsandwich, 2020-06-09:
## Bug report

**Required Info:**

- Operating System:
  - All platforms
- Installation type:
  - Source (ci.ros2.org)
- Version or commit hash:
  - [vcs export --exact](https://gist.githubusercontent.com/nuclearsandwich/3dbdc7d9adf70bb74c9a9a5a985e9d9e/raw/2b9a945170dd9732334500ff0049452f6a27ab36/1231-exact.repos)
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
colcon build --packages-up-to rmw
colcon test --packages-up-to rmw
```

#### Expected behavior

Tests on [test/test_names_and_types.cpp:131](https://github.com/ros2/rmw/blob/b6dcccbb7ec7e1adea6d029868164c5a10910a82/rmw/test/test_names_and_types.cpp#L131) and [test/test_names_and_types.cpp:137](https://github.com/ros2/rmw/blob/b6dcccbb7ec7e1adea6d029868164c5a10910a82/rmw/test/test_names_and_types.cpp#L137) pass.

#### Actual behavior

```

13: [ RUN      ] rmw_names_and_types.rmw_names_and_types_fini
13: /home/jenkins-agent/workspace/nightly_linux-aarch64_debug/ws/src/ros2/rmw/rmw/test/test_names_and_types.cpp:131: Failure
13: Expected equality of these values:
13:   rmw_names_and_types_fini(&names_and_types)
13:     Which is: 0
13:   11
13: /home/jenkins-agent/workspace/nightly_linux-aarch64_debug/ws/src/ros2/rmw/rmw/test/test_names_and_types.cpp:137: Failure
13: Expected equality of these values:
13:   rmw_names_and_types_fini(&names_and_types)
13:     Which is: 0
13:   11
13: [  FAILED  ] rmw_names_and_types.rmw_names_and_types_fini (0 ms)
```

#### Additional information

I was looking at vcs exact output between today and yesterday and I have no inkling as to why these tests which were passing yesterday are failing today. These should be shot for shot tests of code [here](https://github.com/ros2/rmw/blob/b6dcccbb7ec7e1adea6d029868164c5a10910a82/rmw/src/names_and_types.c#L91-L94) and [here](https://github.com/ros2/rmw/blob/b6dcccbb7ec7e1adea6d029868164c5a10910a82/rmw/src/names_and_types.c#L116-L120).

brawner, 2020-06-09:
I'm actually looking at this exact test right now, and I realized it was not implemented correctly.

Related to this test
https://github.com/ros2/rmw/pull/203

brawner, 2020-06-09:
Addressed in https://github.com/ros2/rmw/pull/235

fujitatomoya, 2020-06-10:
I've seen same failure with https://github.com/ros2/ros2/commit/de0a7cf3a76d500df767a8db60236f04b6b1e4c7

nuclearsandwich, 2020-06-16:
Closed by #235

closed 2020-06-16 by nuclearsandwich