ros2/rmw #241: ContentFilteredTopic Support

fujitatomoya, 2020-06-19:
## Feature request

#### Feature description

Draft overview and background is https://github.com/ros2/design/pull/282,

Introduce dds `contentfiltertopic` to `ROS`,  dds specification is [DDS v1.4](https://www.omg.org/spec/DDS/1.4/PDF) 2.2.2.3.3 ContentFilteredTopic Class. ContentFilteredTopic can be created on parent topic and DataReader only receives the messages that matches with filtering expression and expression parameters. If implementation does not support this feature, it returns `not support`.

- rmw implementation support.

| Implementation | Supported | Comment |
| :--- | :--- | :--- |
| RTI Connext | Yes | https://community.rti.com/examples/content-filtered-topic, Not sure if filtering is done on writer or reader side. |
| eProsima Fast-DDS | No |  |
| ADLINK cyclonedds | No |  |
| ADLINK OpenSplice | Yes | http://download.prismtech.com/docs/Vortex/html/ospl/DDSTutorial/topics-etc.html#content-filtering, community version supports filtering on reader side. |

- expected user for this feature.

  - action feedback / status topics.
  - parameter_events topic.
  - user defined topics.

#### Implementation considerations

- As DDS specification, ContentFilteredTopic is not only for DataReader but also DataWriter. But it does not make sense when it comes to DataWriter, since DataWriter knows what it is writing. So it is likely for DataReader to have this feature. (refer to [RTI Connext Sample](https://d2vkrkwbbxbylk.cloudfront.net/sites/default/files/rti-examples/rticonnextdds-examples/examples/connext_dds/content_filtered_topic/c++/cft_subscriber.cxx))
- ContentFilteredTopic requires parent topic, returning capability for ContentFilteredTopic with rmw_create_subscriotion rmw_subscription_t. So that rclcpp/rcl can tell if the implementation supports it or not.
- Based on parent topic and subscription, it can create ContentFilteredTopic with subtopic name, filtering expression and expression parameters. Then implementation creates ContentFilteredTopic and replace the DataReader.
- expression parameter can be changed via setter dynamically, also getter should be provided. but filtering expression cannot be changed unless re-creating ContentFilteringTopic.

Barry-Xu-2018, 2020-10-26:
About implementation estimation.  

1. Implement -- 18 ~ 23 MD
    
   | Module | Estimation (man day) |
   | :-- | :-- |
   |rmw & rmw_implementation |  2 |
   |rmw_connext | 5~10 |
   | rcl  |   4 |
   | rclcpp |   3 |
   | rclpy  | TBD |
   | rmw_fastrtps, rmw_cyclonedds |  4 |
    
3. Integrate and test  --  5 MD
4. Review and modification --  5~10 MD

fujitatomoya, 2020-11-05:
@Barry-Xu-2018 

I think that it would be really nice to break down into some steps, since huge single PR is really heavy work and hard to do review. the following is just idea.

1. rmw ContentFilteredTopic support
  - `is_cft_supported` to ask rmw implementation if ContentFilteredTopic is supported or not.
  - `rmw_create_subscription` with optional fields `const char * filter_expression` and `const rcutils_string_array_t * expression_parameters`
  - `test_rmw_content_filter` to verify ContentFilteredTopic in rmw only.
2. rcl ContentFilteredTopic support
  - basically bind rmw related interfaces to rcl, and verify ContentFilteredTopic in rcl.
  - support internal filtering function on reader side if rmw does not support ContentFilteredTopic.
3. Action Topics with ContentFilteredTopic. (rcl action client mostly)
  - issue ContentFilteredTopic based on current goal ids in action client.
  - additional tests required.
4. Parameter Events with ContentFilteredTopic. 
  - user interface support to set filtering expression and expression parameters. (node name and parameter name)
  - additional tests required.

let's discuss the details offline.

Barry-Xu-2018, 2020-11-05:
@fujitatomoya 

Completely agree.   
Let's discuss it in more detail.

fujitatomoya, 2022-05-10:
https://github.com/ros2/design/pull/282 takes care of all activities content filtered topic. close this issue.

closed 2022-05-10 by fujitatomoya