ros2/rmw #264: Push notification on publisher/subscription match

hidmic, 2020-08-24:
## Feature request

#### Feature description

Currently, the API does not provide (or documents) a generic mechanism to wait for publishers (subscriptions) to match subscriptions (publishers) -- only the current match count is available for query. Thus, code that monitors these counts has to sleep and retry. That's inefficient and potentially brittle. It'd be nice to enable such waits.

#### Implementation considerations

It seems to me that a new pub/sub event would be the simplest solution. All Tier 1 RMW implementations have a form of pub/sub matching listener that can be leveraged:

- eProsima Fast-DDS has it, and both `rmw_fastrtps_cpp` and `rmw_fastrtps_dynamic_cpp` already use it (see [here](https://github.com/ros2/rmw_fastrtps/blob/0959c84f9cca4b2005c9dffe7e18ec71656e5e74/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_publisher_info.hpp#L66) and [here](https://github.com/ros2/rmw_fastrtps/blob/0959c84f9cca4b2005c9dffe7e18ec71656e5e74/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp#L69)).
- RTI Connext has it, and `rmw_connext_cpp` already uses it (see [here](https://github.com/ros2/rmw_connext/blob/2c3f725b6050186e7a215d0cd13634c084e05571/rmw_connext_cpp/src/connext_static_publisher_info.hpp#L61) and [here](https://github.com/ros2/rmw_connext/blob/2c3f725b6050186e7a215d0cd13634c084e05571/rmw_connext_cpp/src/connext_static_subscriber_info.hpp#L57)).
- ADLINK CycloneDDS has it, but `rmw_cyclonedds_cpp` doesn't use it yet (see [here](https://github.com/eclipse-cyclonedds/cyclonedds/blob/623b35d83b9d061187c816f37f4a92798f3b35d1/src/core/ddsc/include/dds/ddsc/dds_public_listener.h#L195-L203)). 

It's worth noting that some [`rcl` tests](https://github.com/ros2/rcl/blob/0980f8837ce76f20c51830b25dd1c8855cec8353/rcl/test/rcl/test_count_matched.cpp#L80-L87) appear to rely on pub/sub matching triggering the graph guard condition. Regardless of that not being true for any of the current Tier 1 RMW implementations, it seems to me that would create too many unwarranted awakes.

ivanpauno, 2020-09-15:
I think that the graph guard condition should be triggered in all graph changes.
A summary of graph changes:
- Anything that can change the output of the functions [here](https://github.com/ros2/rcl/blob/master/rcl/include/rcl/graph.h).
- subscription and publisher matching count: [this](https://github.com/ros2/rcl/blob/c1b3ff72d3d106e322661db3a2702b476eedfe27/rcl/include/rcl/subscription.h#L568) and [this](https://github.com/ros2/rcl/blob/c1b3ff72d3d106e322661db3a2702b476eedfe27/rcl/include/rcl/publisher.h#L603).

About non-dds rmw implementations, FastRTPS didn't use to have wait sets (now FastDDS have).
They implemented waitsets with the use of cv and listeners, so I think every non-DDS rmw implementation should be able to do that.

We could also have more fine grained guard conditions in the future.

About the difference between events and the graph guard condition, I think that in the future we could have a single event/condition type that can be added to the wait set instead of the wait set being able to wait in different kinds of objects.

hidmic, 2020-09-23:
Alright, as per offline discussion, we've agreed that the global guard condition should be able to detect all the aforementioned ROS graph events. https://github.com/ros2/rmw/pull/272 is a first step in this direction, updating the API spec.

hidmic, 2020-09-24:
Alright, since #272 publisher/subscription matching must trigger the graph guard condition. Implementations must now be adapted.