ros2/rmw #321: Inconsistent document with code

IIIIIIIIll, 2022-05-09:
## Bug report

**Required Info:**

- Operating System:
  -  ADE docker (Ubuntu 20.04)
- Installation type:
  - binaries
- Version or commit hash:
  - Foxy 0c23600340f80ceef9d6c74728cde5677e617f15
- DDS implementation:
  - NA
- Client library (if applicable):
  - NA

#### Steps to reproduce issue

https://github.com/ros2/rmw/blob/6d23f9fc6bfe4616e1efeeefee7a927635a83afe/rmw/include/rmw/get_node_info_and_types.h#L140-L177

#### Expected behavior
Consistent documentation with funciton signature. 

#### Actual behavior
Inconsistent documentation with funciton signature. 

#### Additional information
Function contains a boolean value `demangle` in its signature, but the documentation names the same value `no_demangle`. It is pretty confusing what the actual behavior should be when this value is set to `true`.

christophebedard, 2022-05-09:
This was fixed right after Foxy in https://github.com/ros2/rmw/pull/250, which should clear it up: it should be `no_demangle`.

IIIIIIIIll, 2022-05-09:
Thank you for the clarification!

christophebedard, 2022-05-12:
I think this can be closed.

clalancette, 2022-05-12:
Closing with #322