ros2/rmw #330: Connect/Disconnect notification/callback support on publisher

fujitatomoya, 2022-09-13:
## Feature request

#### Feature description

- RMW can notify the event that any subscriptions connect / disconnect on the topic to each publisher.
- Based on this capability, in `rclcpp` and `rclpy`, user application can register user callback that will be called when subscription connects or disconnects on the topic for publisher.

ROS 1 Similar function: https://docs.ros.org/en/noetic/api/roscpp/html/classros_1_1NodeHandle.html#ae4711ef282892176ba145d02f8f45f8d

#### Use case

On visualization node, to suppress unnecessary CPU utilization, enable its visualization process only when RViz is connected.

[image]

#### Related Information / Reference

- [image_pipeline comment out subscription match for temporary](https://github.com/ros-perception/image_pipeline/blob/ea216eba6d2c611adbd78bb9ef9db25816d1405c/depth_image_proc/src/point_cloud_xyz.cpp#L59-L60)
- [ROS2 publisher callback on subscription match](https://answers.ros.org/question/351827/ros2-publisher-callback-on-subscription-match/)
- [pointcloud_to_laserscan does polling to check on subscription match](https://github.com/ros-perception/pointcloud_to_laserscan/blob/rolling/src/pointcloud_to_laserscan_node.cpp#L108)

#### Implementation considerations

Since all tier-1 rmw implementation is DDS implementation, we can support this feature with `on_publication_matched` of `DataWriterListener Interface`. (we confirmed the code of Fast-DDS.)

fujitatomoya, 2022-09-13:
@ivanpauno @wjwwood @asorbini @MiguelCompany @alsora @SteveMacenski @mikeferguson
CC: @ros2/middleware_working_group @Barry-Xu-2018 

This is to open discussion, could you share your thoughts?

ivanpauno, 2022-09-13:
Adding this like a new rmw_event_t seems reasonable to me

mikeferguson, 2022-09-13:
I think this is super useful - at the moment, you can't easily build something like rgbd_launch in ROS2 (where all of the various pipeline elements are lazy publishers/subscribers) - I didn't even realize this wasn't already ticketed somewhere...

I had some notes about use cases (and my current lazy subscriber workaround with a timer that checks get_subscription_count once a second) here: https://www.robotandchisel.com/2020/06/16/ubr1-on-ros2/

alsora, 2022-09-14:
I'm not familiar with the `on_publication_matched` API.
Would it be equivalent to checking `get_subscription_count()` on the publishers objects?
Obviously having a callback would be much better than continuously checking that API, but I just want to make sure that I understand how it would work.

In general this looks a great feature to add.

fujitatomoya, 2022-09-14:
> Would it be equivalent to checking get_subscription_count() on the publishers objects?

the main difference is, application can be notified with event on connect/disconnect.

MiguelCompany, 2022-09-15:
This would be a nice addition, probably as a new `rmw_event_t` as @ivanpauno suggests. We could interpret it as having an event notifying of changes on `get_subscription_count()`

We could do the same for subscriptions, though we should probably have that as a separate issue.

Barry-Xu-2018, 2022-09-22:
We all agreed adding new event.
I submitted a PR to add connect/disconnect event and define `rmw_matched_status_t` for [publisher and subscription event callback](https://github.com/ros2/rclcpp/blob/6a8c61c0269187c9505632647f6a9f5698b523e8/rclcpp/include/rclcpp/qos_event.hpp#L40-L72).    
Please review it.

Barry-Xu-2018, 2022-11-10:
@fujitatomoya   

We should remove `rmw_implementation` in change list since the rmw interface isn't changed in https://github.com/ros2/rmw/pull/331

MiguelCompany, 2022-11-10:
> We should remove `rmw_implementation` in change list since the rmw interface isn't changed in #331

@Barry-Xu-2018 I don't think so, since that repository is there the [tests](https://github.com/ros2/rmw_implementation/tree/rolling/test_rmw_implementation) for this feature should be added.

Barry-Xu-2018, 2022-11-10:
@MiguelCompany 

> I don't think so, since that repository is there the [tests](https://github.com/ros2/rmw_implementation/tree/rolling/test_rmw_implementation) for this feature should be added.

Thank you for your kind reminder.   
Yes. I will check and add tests.

Barry-Xu-2018, 2022-11-11:
In `rmw_implementation`, there is no test related to event currently (Including existing events).   
So I need to add events test file.   
Besides, rmw_implementation test depend on rmw_xxxx (e.g. rmw_fastrtps).   
So I will implement rmw_fastrtps, rmw_connextdds, rmw_cyclonedds at first and then add test to `rmw_implementation`.

fujitatomoya, 2022-11-11:
@Barry-Xu-2018 thanks for checking, i was thinking that there would be some related code or test cases in there, so just make sure that everything we may change on that table at 1st. if we do not need to update, that is also fine, i will remove `rmw_implementaion` from the issue header.

Barry-Xu-2018, 2023-02-15:
Related PRs have been submitted.
- [rmw]
   https://github.com/ros2/rmw/pull/331
- [rmw_implementation]
  https://github.com/ros2/rmw_implementation/pull/216
- [rmw_fastrtps]
  https://github.com/ros2/rmw_fastrtps/pull/645
- [rmw_cyclonedds]
  https://github.com/ros2/rmw_cyclonedds/pull/435
- [rmw_connextdds]
  https://github.com/ros2/rmw_connextdds/pull/101
- [rcl]
  https://github.com/ros2/rcl/pull/1033
- [rclcpp]
  https://github.com/ros2/rclcpp/pull/2105
- [rclpy]
  https://github.com/ros2/rclpy/pull/1083

TODO
- [Demo]
- [ros2_documentation]

fujitatomoya, 2023-02-15:
@Barry-Xu-2018 thanks, i will be reviewing for sure. 

CC: @iuhilnehc-ynos

fujitatomoya, 2023-02-18:
@Barry-Xu-2018 Can we start the CI with all dependent PRs to see all related tests complete w/o error? i will be working on review at the same time.

Barry-Xu-2018, 2023-02-19:
> @Barry-Xu-2018 Can we start the CI with all dependent PRs to see all related tests complete w/o error? i will be working on review at the same time.

Yes. I will run CI in all dependent PRs.

Barry-Xu-2018, 2023-02-21:
@fujitatomoya 

CI on Linux passed. https://github.com/ros2/rmw/pull/331#issuecomment-1436568818  
Next, run CI on Windows.

fujitatomoya, 2023-02-26:
@iuhilnehc-ynos can you do the another review on these PRs?

fujitatomoya, 2023-02-27:
- ros2_documentation
  - Not exactly QoS feature, but events are extended aligned with QoS events.
  - Probably we could extend the docs for https://docs.ros.org/en/rolling/Concepts/About-Quality-of-Service-Settings.html with dedicated section.
- demo
  - after APIs are fixed, we can add the demo codes in the following folders.
  - https://github.com/ros2/demos/tree/rolling/demo_nodes_cpp
  - https://github.com/ros2/demos/tree/rolling/demo_nodes_py

fujitatomoya, 2023-03-03:
@Barry-Xu-2018 can you start the CI with all related PRs?

Barry-Xu-2018, 2023-03-05:
> can you start the CI with all related PRs?

Okay. I will run CI for all related PRs again.

fujitatomoya, 2023-03-16:
@iuhilnehc-ynos @Barry-Xu-2018 once everything is rebased, can you start the CI?

Barry-Xu-2018, 2023-03-17:
@fujitatomoya 

> once everything is rebased, can you start the CI?

Yes. After all are rebased, I will ask @iuhilnehc-ynos help to run CI.

Barry-Xu-2018, 2023-03-18:
@fujitatomoya 

CI was done.  
https://github.com/ros2/rmw/pull/331#issuecomment-1473429709   

The failed test cases on Windows are unrelated to my change.

mjcarroll, 2023-03-22:
Everything up to `rclcpp`/`rclpy` have been approved and merged.  I'm going to leave this to you to close when demos/documentation are updated.  Thanks for all the hard work and iteration!

fujitatomoya, 2023-03-22:
@mjcarroll appreciate it for taking care of those PRs 👍 thanks a lot 🙇‍♂️ (ありがとう)

fujitatomoya, 2023-03-22:
@Barry-Xu-2018 

the rest will be https://github.com/ros2/rmw/issues/330#issuecomment-1447239439, can you also address these?

Barry-Xu-2018, 2023-03-24:
Demo PR was submitted.  ros2/demos#607  

About the document, I didn't find a detailed description of the "event" in the document. Only a brief introduction is provided, and there are no examples given.
Based on this file, I add brief introduction for matched as below. (New added event `Inconsistent topic` isn't described. I add brief introduction)
https://github.com/Barry-Xu-2018/ros2_documentation/commit/c183c93349a9551bd310d0af976d6ad1470da5ad

fujitatomoya, 2023-03-24:
> Based on this file, I add brief introduction for matched as below. (New added event Inconsistent topic isn't described. I add brief introduction)

I think that would be better to add an introduction like this, i left some comments on it, can you make the PR?

and, probably doc for inconsistent topic, we would want to create another PR.

Barry-Xu-2018, 2023-03-26:
> I think that would be better to add an introduction like this, i left some comments on it, can you make the PR?
>
> and, probably doc for inconsistent topic, we would want to create another PR.

Okay. I created a PR ros2/ros2_documentation#3406.

fujitatomoya, 2023-04-12:
https://github.com/ros2/rmw_fastrtps/pull/683 has been merged.

fujitatomoya, 2023-04-13:
complete all related PRs, see https://github.com/ros2/rmw/issues/330#issue-1371797574#353 

i will go ahead to close this issue.

closed 2023-04-13 by fujitatomoya

fujitatomoya, 2024-03-11:
@Ryanf55 

> Is this feature possible to backport to humble?

unfortunately I do not think so, originally we did not consider the backport of this feature. this adds many APIs including RMW interfaces.

> If not, is there any other approach to verify all topics are matched before starting a test when using rclpy?

we can check if corresponding publishers or subscribers on the topic, which uses `NodeGraph`.

https://github.com/ros2/rclpy/blob/f1873bfb1e3261e617cb0a4c9cfb6ddcf57b6747/rclpy/rclpy/node.py#L2124-L2152

i think this could be used to check if `someone` is on the other side, but downside is that this is not an `event`, so we need to do polling to ask if there is `someone` on the topic.

Note: with `rclcpp` we can have an event from [NodeGraph](https://docs.ros.org/en/rolling/p/rclcpp/generated/classrclcpp_1_1node__interfaces_1_1NodeGraph.html) guard condition update,  but this does not tell you what exactly has been updated. after all, we need to check how graph is updated from application.

that is pretty much that i know, hopefully this helps.

tonynajjar, 2024-05-15:
This is a really cool feature but it feels quite verbose for the very common use case: "publish only if there are subscribers". For that I still prefer something as simple as:

```c++
  if (publisher_->get_subscription_count() > 0) {
    /* heavy work for visualization
    */
    publisher_->publish(msg);
  }
```

It was mentioned that the drawback with this approach is the need to constantly poll with an external timer but in many cases this block of code would already be in a timer/subscriber callback (you can see many examples in Nav2). So apart from this drawback, is there anything else matched events are better with? Thanks

tonynajjar, 2024-05-15:
Yes I know it exists, thanks for mentioning the caveat though. I was just asking if this method has any other known drawbacks compared to matched events + counting subscribers manually (which would be a much more verbose solution for this simple yet very common use case)

fujitatomoya, 2024-05-15:
@tonynajjar the original motivation is, we need to create the data pipeline dynamically only if it comes necessary for network and cpu efficiency as we do develop edge devices. having polling mechanism and calling API to query the network connectivity is not really cost effective cz we have to do that every single time and most likely the network connectivity has not changed yet... i think it all comes down to the use case. (we sometimes use `get_count` as well.)

something i want to mention about the drawback (maybe not drawback but more like feature) as comparison is, that the matched event is synchronized information, that guarantees how the mached endpoits changes.

for example, publisher online -> get_sub_count called and 0 returns -> subscriber online -> subscriber offline -> get_sub_count called and 0 returns, this case it appears that nothing has happened in the network connectivity. but it actually did. the same thing could happen that publisher can see the same 1 count of subscription, but this subscription could be a different one in between `get_count` API calls. with matched events, we can see those changes for sure, so that could be one of the difference too.

@Barry-Xu-2018 please correct me if i got anything wrong above.

tonynajjar, 2024-05-15:
I see your point about not missing any information in between API calls. Thanks everyone for the clarifications, hopefully that helps more people in the future.