ros2/rmw #356: Extend `names_and_types` to contain type hashes

emersonknapp, 2023-07-26:
## Feature request

#### Feature description

REP-2016 type hashes now exist and are passed during discovery for the 3 DDS implementations, but are only exposed through `rmw_topic_endpoint_info_t`, which is given by `get_publishers_info_by_topic`/`get_subscriptions_info_by_topic`, but otherwise is not exposed.

When querying information about a _node_ rather than a _topic_, one generally uses `rmw_get_publisher_names_and_types_by_node`(etc) which returns the topic name and typename, but does have access to the type hash, which it would be appropriate to return in these functions, making it easier to then query information about those topics.

#### Implementation considerations

This probably looks like:

```
typedef struct RMW_PUBLIC_TYPE rmw_names_and_types_s
{
  /// Array of names
  rcutils_string_array_t names;

  /// Dynamic array of arrays of type names, with the same length as `names`
  rcutils_string_array_t * types;

  /// NEW FIELD
  /// Dynamic array of type hashes, with the same length as `names`
  rosidl_type_hash_t * type_hashes;
} rmw_names_and_types_t;

```

And then will require filling out this information in all `names_and_types` APIs.

This would be ABI-breaking, so probably can only be planned for Jazzy.

wjwwood, 2023-07-27:
Makes sense to me, basically anywhere we have the type of a topic we should have the hash.

fujitatomoya, 2023-07-28:
sounds good to me.

asancho99, 2024-12-10:
I need the hash to obtain the information of an interface in runtime using the GetTypeDescription service. With the type hash discovery for msg is really easy since just executing the "ros2 topic info -v /topic" I obtain the required hash to call the service. But to use the GetTypeDescription with actions and services I need to check the hash manually in the install folder, so it is only possible if you have access to the source code. It could be really interesting to include this feature request to discover the hash for actions and services to be able to take the advantage of knowing the structure of the interfaces in runtime (without access to the code) using the GetTypeDescription service?

gramss, 2025-08-22:
https://github.com/siemens/rosita would also benefit from this function.

If this is not picked up by other maintainers, we could try building a PR on our own. Although, guidance from existing maintainers would be highly welcome. The description by @emersonknapp could be already all needed information to start this. Still, happy to connect and get some pitfalls out of the way, before we start this effort on our site.