ros2/rmw #363: Interface for a function that returns the serialized size of a message

kfabian, 2024-01-15:
## Feature request

#### Feature description
I noticed that none of the major rmw implementations implements `rmw_get_serialized_message_size`. I wonder if that is the case because it works on the `rosidl_runtime_c__Sequence__bound` struct and not an actual instance of a message. I could not really grasp how the `rosidl_runtime_c__Sequence__bound` should be used and to what structure `* data` should be pointing to.

#### Implementation considerations
Maybe it would be helpful to add a function interface that works on an actual instance of a message, that should be more straightforward to implement, e.g. for cyclonedds similar to [this](https://github.com/ros2/rmw_cyclonedds/blob/f7f67bdef82f59558366aa6ce94ef9af3c5ab569/rmw_cyclonedds_cpp/src/rmw_node.cpp#L1849) code.

clalancette, 2024-01-16:
Hm, the fact that none of them implement it means the API should probably be deprecated and removed.  I think this was part of an early attempt to be able to send preallocated messages, but never got implemented.  @mjcarroll can you comment here?

mjcarroll, 2024-01-19:
I think that is the origin of it.  I think that the loaned message APIs came after and superceded this concept.  I think we can safely deprecate and remove the existing APIS.

Second, I think that the idea of working on a message instance may work as well.  The `Sequence__bound` structure was intended to provide bounds for unbounded sequence fields so that allocation size could be correctly computed.