ros2/rmw #401: direct deserialization of an ros2 msg based on a sring.

Tanishq30052002, 2025-04-07:
### Description

Currently, it requires type support and the implementation of a dynamic library from the user's side in case they need the deserialized message. why not have a function that deserialises the serialized message without bothering about all that stuff?

### Motivation

Currently, it requires type support and the implementation of a dynamic library from the user's side in case they need the deserialized message. why not have a function that deserialises the serialized message without bothering about all that stuff?

Tanishq30052002, 2025-04-10:
Hi @ahcorde @wjwwood 

Can you please share your views on this?

fujitatomoya, 2025-04-11:
@Tanishq30052002 can you describe more details about how you want to change the user behavior with example codes?

for user-perspective, what/which APIs or interfaces are you talking about here or we need to introduce? it would be really helpful to understand if we have examples that tell us what the problem is and how we want to change/introduce.

Tanishq30052002, 2025-04-11:
Let me provide a glimpse of what I was trying to do so it is clearer what the aim was.

I created a generic subscriber where any message can be published, and our goal was to print the information from that message in a human-readable format.

Challenges:
* I did not know what kind of message it was.
* I can not import all the headers for every case of the msg type.

Approach Taken:
* I get to know what kind of message it is from get topic type whenever a msg it starts getting published. But here, the msg type is in the string format. So I browsed over certain solutions and found there is a method of deserialization of a ROS2 msg directly. For the same, I need to create an object of the Serialization class with type support, and it has a function that finally calls the rmw_deserialize.

* Snippet of that function:
```
/// Deserialize a ROS message.
/**
 * The given rmw_serialized_message_t's internal byte stream buffer is deserialized
 * into the given ROS message.
 * The serialization format expected in the rmw_serialized_message_t depends on the
 * underlying implementation.
 *
 * \pre Given serialized message must be a valid non-null instance, such
 *   as that returned by `rmw_serialize()`, matching provided typesupport
 *   and ROS message.
 * \pre Given typesupport must be a valid non-null instance, as provided
 *   by `rosidl` APIs.
 * \pre Given ROS message must be a valid non-null instance, initialized
 *   by the caller.
 *
 * <hr>
 * Attribute          | Adherence
 * ------------------ | -------------
 * Allocates Memory   | Maybe [1]
 * Thread-Safe        | No
 * Uses Atomics       | Maybe [2]
 * Lock-Free          | Maybe [2]
 * <i>[1] if the given ROS message contains unbounded fields</i>
 * <i>[2] rmw implementation defined, check the implementation documentation</i>
 *
 * \param[in] serialized_message the serialized message holding the byte stream
 * \param[in] type_support the typesupport for the typed ros message
 * \param[out] ros_message destination for the deserialized ROS message
 * \return `RMW_RET_OK` if successful, or
 * \return `RMW_RET_BAD_ALLOC` if memory allocation failed, or
 * \return `RMW_RET_ERROR` if an unexpected error occurs.
 */
RMW_PUBLIC
RMW_WARN_UNUSED
rmw_ret_t
rmw_deserialize(
  const rmw_serialized_message_t * serialized_message,
  const rosidl_message_type_support_t * type_support,
  void * ros_message);
```

* Also, typesupport is supposed to be found out dynamically.

* So, my idea was why not to have an overload function to which we can provide serialized msg and string for message type and which can return us the deserialized msg.

christophebedard, 2025-04-11:
You can get the typesupport struct from a message type name string using [`get_message_typesupport_handle()`](https://github.com/ros2/rclcpp/blob/c01602f7a68f25e96bbb7719e874d88dd2018d88/rclcpp/include/rclcpp/typesupport_helpers.hpp#L54). There are similar functions for services and actions.

Then you can deserialize the message.

Tanishq30052002, 2025-04-12:
@christophebedard 
I tried doing it the way you suggested and stuck with the error attached below, can you please help me in getting out of this
[image]

Tanishq30052002, 2025-04-12:
@christophebedard In ros2 humble this feature is currently missing

fujitatomoya, 2025-04-12:
> In ros2 humble this feature is currently missing, it is just specific to msgs.

humble also supports this feature for msgs.

https://github.com/ros2/rclcpp/blob/humble/rclcpp/include/rclcpp/typesupport_helpers.hpp#L40-L53

for services (supported by jazzy and rolling) and actions (supported by only rolling and next kilted), those are new features required by introspection and generic endpoint support for rosbag2.
since those are new features, i do not think we can backport this.

Tanishq30052002, 2025-04-12:
Any special reason why we cannot backport this?
Since Humble is currently in use and will work until 2027, should we add support for services and action?

closed 2025-04-12 by Tanishq30052002